ChimeraTK/DeviceAccess #2: Compile error with gcc 6.2/clang 3.8.1

jmrahm, 2017-01-12:
Hey guy's,

I tried to compile your DeviceAccess with the newer compiler gcc 6.2 and clang 3.8.1 and I get an semantically error in:

`if(_deviceBackendPointer` != false) {        // Some backend is assigned: close it if opened.

Message:
> DeviceAccess/device/src/Device.cc:17:30:
> error: no match for ‘operator!=’ (operand types are ‘boost::shared_ptr<mtca4u::DeviceBackend>’ and ‘bool’)

I need especially  the clang compiler to run some static code analysis in our OPC UA Adapter. I would be glad if you can fix it. ;-)
Thanks.

soerengrunewald, 2017-01-12:
I face the same issue, which boost version have you installed. I have v1.60 installed.

jmrahm, 2017-01-12:
I have v1.61.

mhier, 2017-01-12:
The error message seems to be correct, at least newer boost versions define the conversion operator to bool as explicit. The strange thing is that gcc doesn't complain on Ubuntu 16.04, which uses boost 1.58, which also defines this operator as explicit already (only very old versions didn't).

I fixed it in the current master branch, but I cannot be certain if there are more errors of this type. Can you please test it?

jmrahm, 2017-01-12:
Perfectly, the first comparison with "!=" works well. ;-)
It's my misstake, above I present to you only the first error. But there are two more with the same cause in:

> DeviceAccess_src/src/external-ChimeraTK-DeviceAccess/device/src/Device.cc:287:30:` error: no match for ‘operator!=’ (operand types are ‘const boost::shared_ptr<mtca4u::DeviceBackend>’ and ‘bool’)

> DeviceAccess_src/src/external-ChimeraTK-DeviceAccess/device/src/Device.cc:224:32:` error: no match for ‘operator==’ (operand types are ‘const boost::shared_ptr<mtca4u::DeviceBackend>’ and ‘bool’)

soerengrunewald, 2017-01-12:
This is a bit wired, cause the cast to bool is obviously supported by the class, but not used or recognized by the compiler. May this is caused by the global operators, but they are clearly for smart_ptr compare.
Anyway, I would suggest to simply replace the (ptr != false) and (ptr == false) with if (ptr) and if (!ptr), this worked for me.

mhier, 2017-01-12:
I fixed the other two comparisons. I hope there are not more in other files, I feel a bit blind with my old compiler ;-)

jmrahm, 2017-01-12:
Sry, but now there are some other errors. But after these little fixes it work very well for me. ;-)

The first one: 
> DeviceAccess/tests/unitTestsNotUnderCtest/testParserUtilities.cpp:63:15:`
 error: reference to ‘utils’ is ambiguous
... and many more

```
BOOST_CHECK(utils::concatenatePaths("a/", "/b") == "/b");
DeviceAccess/tests/unitTestsNotUnderCtest/testParserUtilities.cpp:5:42: note: candidates are: namespace utils = mtca4u::mtca4u::parserUtilities;
```
**Fix:** Rename "utils" to something else. it seems to be already use by boost.

The second one:
> DeviceAccess/tests/executables_src/testDMapFilesParser.cpp:297:3: error: expected ‘;’ before ‘try’

> DeviceAccess/tests/executables_src/testDMapFilesParser.cpp:330:3: error: expected ‘;’ before ‘try’

> DeviceAccess/tests/executables_src/testDMapFilesParser.cpp:375:3: error: expected ‘;’ before ‘try’

**Fix** Set a ";" before try ;-)

Thats all. Thank you!

mhier, 2017-02-23:
I just fixed those issues. DeviceAccess now compiles on gcc 6.3 (openSuSE Tumbleweed), although the tests crash after finishing (will be investigated separately).

closed 2017-02-23 by mhier