ChimeraTK/DeviceAccess #4: Compile error -> missing file

jmrahm, 2017-03-23:
Hey guys,

I'm not able to compile the project, because a file is missing:
DeviceAccess/examples/custom_backend_registration/CustomBackend.cc:1:121: fatal error: mtca4u/DummyBackend.h: No such file or directory
 #include <mtca4u/DummyBackend.h> // Would probably be DeviceBackendImpl or NumericAddressedBackend in a real application

Many thanks in advance

killenb, 2017-03-23:
Hello Julian,

I cannot reproduce this problem. Please try deleting the build directory and rerun cmake.

Cheers
Martin

jmrahm, 2017-03-23:
I already tried. 
The interesting part is both files exit, but the folder structure of the header file is not like the include directive. Nevertheless I delete "mtca4u/" from every include, so it works for me. 

Greetings
Julian

killenb, 2017-03-23:
Which distribution, especially cmake version are you using?
Where do you have your build directory? (We usually use a directory named build in the root of the checkout, so it is only kind-of out of source, but I also tried real out of source in directory next to the one with the sources). I also tried clang, which gives tons of warnings but runs through.

The idea of the examples is that they should work with an installed version of the library, so we are copying the headers to ${builddirectory}/include/mtca4u and add this to the include path. If you remove the namespace directory the example only builds within the source code, but not stand alone against an installed version of the library. It would be good to understand why this is happening.

mhier, 2017-03-23:
Maybe you can also provide us with the output of "make VERBOSE=1" containing the error message and the command leading to the error?

mhier, 2017-03-23:
I could reproduce the error and I think I have found a fix. The problem was that the header files were copied to the build directory sometimes too late, because the dependencies were not correct. I changed the CMakeLists.txt in a way that the header files are now copied already during the configuration step (i.e. when running cmake itself). Can you check if its working for you now?

jmrahm, 2017-03-24:
Hey,
sry for the late response. And yes, now its compiles well.
Tank you!

closed 2017-03-24 by jmrahm