ChimeraTK/DeviceAccess #9: clang on linux is failing

killenb, 2017-08-16:
The decision in CMakeLists.txt to change the linker flags based on the compiler is not working.
`-Wl,$<$<CXX_COMPILER_ID:Clang>:-all_load>$<$<NOT:$<CXX_COMPILER_ID:Clang>>:--whole-archive>`
Under linux the linker still is ld, even if clang is used as compiler. I think the decision should be based on the OS, not the compiler.

Please make it compile with clang under linux by using `CXX=clang++ CC=clang cmake ..` when configuring cmake. I cannot test it without potentially breaking it on MAC because I don't know how to make the decision for that OS.

smarsching, 2017-08-17:
You are right. When using Clang on Linux, the GCC linker is used by default. Even when explicitly using the LLVM linker on Linux, it does not accept the options, so it seems like they are specific to the macOS version of the LLVM linker. I will change it so that the `-all_load` option is only used on macOS.

smarsching, 2017-08-17:
I committed a patch that fixed the issue on Linux and still allows to build on macOS.

closed 2017-08-17 by smarsching