ChimeraTK/DeviceAccess #18: Explicitly linking to librt causes linker-problems on non-Linux systems

smarsching, 2018-10-22:
When trying to build DeviceAccess on macOS (10.13.6) I have the problem that the linker complains about `-lrt` not being found. Of course, the linker is right: There is no such library on macOS.

The question is: Why do we specify this library as part of the `target_link_libraries` in `CMakeLists.txt`? I believe that librt is part of glibc and thus there should be no need of explicitly linking to this library on Linux (the linker should automatically include it anyway).

If it is indeed necessary to explicitly link to this library on Linux, we have to find a way to only add it to the `target_link_libraries` when building on Linux.

killenb, 2018-10-22:
Unfortunately this seems to be different on Linux and MacOS. The compiler complains about a missing symbol without this flag. 
```
/usr/bin/ld: CMakeFiles/testSharedDummyBackendExt.dir/unitTestsNotUnderCtest/testSharedDummyBackendExt.cpp.o: undefined reference to symbol 'shm_open@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/librt.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

```
I found a proposal on stackoverflow to use a system-specific link option, which I put it.
https://stackoverflow.com/questions/7695638/conditional-cmake-link-to-rt-library 
(I don't know if it's Linux or MacOS which does it differently from the rest, so I kept the example. Works for the two flavours we have at the moment).

Please try the HEAD of master if it now works for you.

smarsching, 2018-10-22:
Your patch almost fixed the problem for me. I say almost, because the same problem existed in `tests/CMakeLists.txt`, so I copied your workaround to that file as well and now it is fixed for me (I already commited this change to the master branch).

Regarding the affected systems, I found something interesting on StackOverflow: https://stackoverflow.com/questions/6754032/

It seems like there some versions of Linux, that do not provide libpthread and librt either. Don't be fooled about the talk of "Bionic" in that StackOverflow thread. In this context, this is not Ubuntu 18.04 LTS, but the C library used by Android.

Still, this means that simply checking for macOS or Linux is probably not sufficient - at least if we do want to support all flavors of Linux. I can imagine that some embedded Linux distributions may use a different implementation of libc (than the GNU C library) as well.

The best way of solving this would probably be to check whether the system supplies libpthread and librt and only linking to them if they are available.

phako, 2019-02-25:
> Still, this means that simply checking for macOS or Linux is probably not sufficient - at least if we do want to support all flavors of Linux. I can imagine that some embedded Linux distributions may use a different implementation of libc (than the GNU C library) as well.

That's definitly not sufficient, it fails on OpenBSD now. The -lrt dependency is dragged in by boost - Ideally boost on the platform should now whether or not it needs -lrt

phako, 2019-02-25:
Ok, boost isn't being helpful here:
Building Boost.Interprocess

There is no need to compile Boost.Interprocess, since it's a header only library. Just include your Boost header directory in your compiler include path.

Boost.Interprocess depends on Boost.DateTime, which needs separate compilation. However, the subset used by Boost.Interprocess does not need any separate compilation so the user can define BOOST_DATE_TIME_NO_LIB to avoid Boost from trying to automatically link the Boost.DateTime.

> In POSIX systems, Boost.Interprocess uses pthread system calls to implement classes like mutexes, condition variables, etc... In some operating systems, these POSIX calls are implemented in separate libraries that are not automatically linked by the compiler. For example, in some Linux systems POSIX pthread functions are implemented in librt.a library, so you might need to add that library when linking an executable or shared library that uses Boost.Interprocess. If you obtain linking errors related to those pthread functions, please revise your system's documentation to know which library implements them. 

(from [Boost::Interprocess documentation](https://www.boost.org/doc/libs/1_67_0/doc/html/interprocess.html#interprocess.intro.introduction_building_interprocess))

closed 2019-02-25 by phako

smarsching, 2019-02-25:
@phako: Thanks for fixing this properly!

commit def2cd17 (2019-02-25): Add check for shm_open

Explicitly check whether we need to link to -lrt for that or not. Also,
use CMake's thread library detection to link against the proper
threading library if necessary

Fixes #18