ChimeraTK/DeviceAccess #27: Plugin loading and binary incompatibility - part 1

mhier, 2019-02-08:
(Writing is not yet complete...)

Background:

The current plugin loading mechanism in the

nshehz, 2019-03-04:
We used libraries from https://github.com/mhier/segregatedLinkingExample (by Martin Hierholzer),
where LibA and LibB are binary incompatible and LibB depends on LibC.
- Linked DoocsBackend against LibA.
- Linked QtHardMon against LibC.
- Used QtHardMon to open a doocsBackend. 
 
Loaded DoocsBackend in DeviceAccess using **dlopen** and noticed that the function called from within the DoocsBackend to LibA was not correct. (It called LibC function).

Loaded DoocsBackend in DeviceAccess using **dlmopen** and noticed that the function called from within the DoocsBackend to LibA was correct.

mhier, 2019-03-05:
#38 is the pull request for this ticket. Review and bug fixing in progress.

mhier, 2019-03-07:
New tickets will be created for this issue.

closed 2019-03-07 by mhier