ChimeraTK/DeviceAccess #30: Logical Name Mapper: Write plugin for double buffering

mhier, 2019-02-13:
Depending on #29 

Write a register plugin (or how this is now called, see #29) which handles double buffering of read transfers.

The plugin has 3 parameters:
- Name of the second buffer (first buffer is assumed to be the target register)
- Name of the control register which enables/disables buffer switching in firmware (0: disabled, 1: enabled - written by the plugin to the firmware)
- Name of the status register which defines the buffer *not* to read (read by the plugin from the firmware), i.e. 0 means read second buffer, 1 means read first buffer.

The plugin should follow the following procedure when reading:
- in doPreRead(): Set the control register to 0
- in doReadTransfer() etc:
  - Read the status register
  - Read the corresponding buffer determined by the status register (with the inverted logic as mentioned above)
- in doPostRead(): Set the control register to 1

Writing shall be disallowed - all registers with this plugin become read-only.

The plugin should be working together with plain scalars and 1D arrays as well as extracted channels of 2D registers, even inside a TransferGroup. This should already be the case when following the above procedure (there will be unnecessary writes to the control register, but this is acceptable). Extracted channels will need special attention, since the accessor for second buffer must be created in the same way as the one of the first buffer - as an extracted channel with the right parameters.

Definition of done:
- [ ] Implementation
- [ ] Tests with different register types and with and without TransferGroup
- [ ] Short Doxygen reference documentation of the plugin
- [ ] Short example how to use it on the Doxygen tutorial page for the logical name mapping

ckampm, 2020-09-15:
Documentation of the FW interface can be found here:
https://redmine.msktools.desy.de/projects/utca_firmware_framework/wiki/DAQ_-_Data_Acquisiton_Module

The control register is `WORD_DUB_BUF_ENA`. The status is given by `WORD_DUB_BUF_CURR`, which holds the buffer number written by the FW, i.e. as required above, the buffer not to read by the plugin.

jhktimm, 2020-10-08:
See follow up ticket #189

closed 2020-10-08 by jhktimm