ChimeraTK/DeviceAccess #33: DummyBackend: allow writing to read-only registers for debugging

mhier, 2019-02-20:
The DummyBackend (and the derived SharedMemoryDummyBackend) is frequently used for testing. This usually involves simulating firmware behaviour. When not writing a specific dummy which simulates the FW behaviour, one needs to write to the registers which are normally set by the FW e.g. from the test routine. Those registers commonly are flagged as RO and hence cannot be written, which prevents then using the dummy for tests.

The general behaviour should not change, it may be important for the tests that RO registers are really not writeable and writing throws an error. To still allow changing those registers, a trick shall be introduced: Each read-only register should have a writeable companion, which points to the same address.

To not disturb applications which rely on the catalogue, the catalogue should not show these companion registers. The user has to know they exist or simply try it out. They should be named like the actual register but with the suffix ".DUMMY_WRITEABLE", e.g. for the register

APP.0.WORD_STATUS

a register called 

APP.0.WORD_STATUS.DUMMY_WRITEABLE

would be created.

QtHardMon also needs to be changed, as it needs to test for these registers and offer writing to them. A separate ticket will be created.

Definition of done:
- [ ] Implementation
- [ ] Tests: registers not visible in catalogue, registers can be written and values can be retrieved through the main register, etc.
- [ ] Documentation: Create a short page with the title "Testing applications with the DummyBackend", write a short stub, then add documentation for this new feature.
- [ ] Ask ticket master to select the QtHardMon ticket for the board

killenb, 2019-02-20:
How about having a new "main directory" DUMMY_WRITEABLE which then replicates the main tree? So the register would be
DUMMY_WRITEABLE.APP.0.WORD_STATUS
Like this the registers could also be available in the catalogue and would not clutter the main directories. QtHardMon could stay as is (the dummy registers are marked writable in the catalogue, the others are r/o). In fact, one could, just out of convenience, replicate all registers, not only the read only in the dummy folder

mhier, 2019-02-20:
I would not change the catalogue. Applications, like ApplicationCore-based servers, might connect even the complete device with the control system.

ckampm, 2019-07-03:
Conclusion: The catalogue is not changed and the special registers use the proposed suffix ".DUMMY_WRITEABLE".

closed 2019-07-05 by ckampm