ChimeraTK/DeviceAccess #37: Multiplex Sequence DAQ Region should not be in Register Catalog

cagil-gumus, 2019-02-26:
This low priority issue causes QtHardMon to display phantom registers that are not accessible which lowers the user experience.

phako, 2019-05-03:
This is more complex as it looks.

Apart from a couple of tests Relying on the information being available in the register catalogue, the main blocking issue is that the NumericAddressedBackendMuxedRegisterAccessor also uses the information from the catalogue to deduce the information necessary to bootstrap itself.

This leads to the question on what the catalogue is supposed to be and needs a design phase in what we want it to be and how to transport the information otherwise, if necessary

phako, 2019-05-14:
Marked as design ticket

mhier, 2019-06-06:
This is not an umbrella ticket, but it depends on the umbrella #41.