ChimeraTK/DeviceAccess #46: LogicalNameMapper accessor plugin: apply bit mask

mhier, 2019-03-15:
Write an accessor plugin for the LogicalNameMapper (see #29) which applies a bit mask to the data. The bit mask is specified as a parameter to the plugin. The data which is read or written through the accessor should be bit-wise ANDed with the bit mask before passing on.

DoD:

- [ ] Implementation
- [ ] Test
- [ ] User documentation

killenb, 2019-03-18:
For writing a simple AND-mask is not sufficient. What is needed is a "read modify write" which is protected by a lock per target register for thread safety.

```
{
std::lock_guard<std::mutex> lock(thisTargetsMutex);
targetValue.read()
targetValue =  (targetValue & ~userMask) |  (userValue & userMark)
targetValue.write()
}

```

mhier, 2019-03-18:
Yes, but instead of the blocking read, readLatest() should be called, i.e.:
```
{
  std::lock_guard<std::mutex> lock(thisTargetsMutex);
  targetValue.readLatest()
  targetValue =  (targetValue & ~userMask) |  (userValue & userMark)
  targetValue.write()
}
```

killenb, 2019-03-18:
Comment to locking read-modify-write: A first implementation of the plugin could be read-only.

killenb, 2019-03-18:
Possible extension: Extract a sub-word out of the register. For instance bits 0..5 are a bit-field with flags, bits 8..23 represent 16 bit signed with 3 fractional bits (needs mask, bit-shift and fixed-point conversion).

Would one rather have 3 nested plugins for this?