ChimeraTK/DeviceAccess #51: implement push type elements in dummy backend

jhktimm, 2019-03-20:
NumericAddressedBackend need to have push type element support.
Map file design have to be match for this.

This is needed for an example for the DataConsistencyGroup and may be an example for ReadAnyGroup.

mhier, 2019-06-06:
It is also required to support PCIe interrupts etc.

killenb, 2019-06-06:
The push type feature probably has three components:
* An accessor with a future_queue. All accessors which are handed out are registered at the backend.
* A dispatcher which pushes into the accessors. This is needed because there might be multiple accessors for one "register", so the data is fanned out here. In case of interrupt handlers the data might be interpreted (interrupt number is send and there is one "register" for each interrupt type). Usually this kind of dispatcher is implemented as a thread.
* A blocking function which the dispatcher thread is waiting for.

Obviously the blocking function is highly backend dependent. But also the dispatcher does not have much common functionality to use. In case of a dummy backend there probably is a dispatcher without  a blocking function with a waiting thread, because the sending action is done directly by the test. Whether there is one list of all accessors or a list for each "register" depends on the backend, so there is probably not much code to reuse.

The only common thing is the accessor. A first ticket could be to implement an AsyncNDRegisterAccessor, which in contrast to the SyncNDRegisterAccessor is the implementation being used. One does not have to derrive from it. 

In the uio branch there is an implementation with such an accessor. It turned out that unregistering an accessor needs to be solved in the interface. The dispatcher must not write to it if the instance has gone. Have a look at this branch. It should use the functionality resulting from this ticket.

Another example for a push type implementation is the DOOCS backend. This should also be studied to see how a working solution looks like.

Foreseen users of this feature:
* uio backend
* RadCon backend
* DummyBackend

killenb, 2021-03-16:
This feature has been moved to the DESY MSK tracker https://redmine.msktools.desy.de/issues/7874 (sorry, DESY internal access only)

closed 2021-03-16 by killenb