ChimeraTK/DeviceAccess #61: New requirement for all backends: do not close device if runtime error occurs

mhier, 2019-04-29:
As a consequence of the recent developments for exception handling in ApplicationCore, a problem in our current practise for backend implementation was found: At the moment, most backends will go to the closed state of the backend when they throw a ChimeraTK::runtime_error. Since a ChimeraTK::logic_error is thrown when read/write operations are executed on closed backends, this conflicts in multi-threaded environments with the definitions of the exceptions: An application which executes concurrent read/write operations on the same device will see a logic_error in all other threads after it has received a runtime_error in one thread, because then the device is closed. It is impossible to detect this reliably - calling isOpen() prior to any read/write operation is not sufficient as the exception might occur in between the two calls in another thread.

Hence, all (!) backends need to be carefully checked. No backend must go to its closed state unless explicitly closed by the application by calling close(). Please check that neither close() is called nor the flag is changed directly at any point inside the implementation.

killenb, 2020-01-14:
In version 01.09 tests have been implemented in DeviceAccess that `open()` can be called again without throwing. The DoocsBackend only sets the _opened flag and exits, it cannot throw.
For the UIO backend a follow-up ticket has been done (#99).

mhier, 2020-09-11:
Can we close this one now?

vargheseg, 2020-09-11:
The rebot backend should be good.

closed 2022-12-21 by mhier