ChimeraTK/DeviceAccess #70: Implement recovery scheme for PCIe backend

killenb, 2019-08-06:
depends on #69 

* When a runtime_error is thrown, the _hasErrors flag shall be set to true
  * Make sure it is cleared when needed, for instance when closing the backend
* When calling open() again, the backend shall close the device node and try to re-open it.
  If this is successful, _hasErrors is reset, otherwise a runtime_error is thrown.

killenb, 2020-01-13:
FIXME: `_hasError` shall not be used. It will be removed, together with the default implementation of the `isFunctional()`.
Overload `isFunctional()` instead. If needed you can cache the error state in a variable, but you could also check if it device is working differently (maybe introduce a reference register with reference register that can be used).

killenb, 2020-01-13:
Tests can also be executed with real hardware, and instructions how to do a test sequence for reference/repeatability,

killenb, 2020-01-15:
This ticket is correlated with #97 (remove isFunctional default implementation).

mhier, 2020-01-20:
I have added a test in the corresponding wip branch:
tests/unitTestsNotUnderCtest/testPcieErrorHandling.cpp

@killenb: Can you please have a look at it if this test would fulfil already your DoD?

The actual implementation is still ongoing.

mhier, 2020-01-21:
Implementation done, but slightly different as suggested in the ticket. `isFunctional()` now tests whether the device is present by reading byte 0 of bar 0.

A manual test has been written and run a couple of times on actual test hardware. Recovery seems to work even without changing the driver. In real life (e.g. with interrupted DMA transfers) additional problems might occur though which are difficult to test.

closed 2020-01-21 by mhier