ChimeraTK/DeviceAccess #71: Extend the map file to be able to describe Push-Type registers

killenb, 2019-08-22:
Currently registers in mapped devices are implicitly poll-type. Some backends (uio, pcie) are able to deliver interrupts. This has to be integrated into the map files. Maybe the RegisterInfo object should be extended.

* [ ] Propose a possible extension (or multiple) and have it (them) reviewed before implementing
* [ ] Extend the RegisterinfoMap::RegisterInfo and the map file parser
* [ ] Write tests for the new functionality

Afterwards it should be adapted in the uio branch so it can be merged into the master.

phako, 2019-08-22:
We can have two approaches here let PCIE/UIO drive the format or be more generic.

### PCIE/UIO POV

Extend the data interpretation field with a INTERRUPT information. The width would denote the maximum bitmask, it could also carry an optional bitmask of relevant bits

Optionally a number of none-type variables mapped to the bits of the variable defined as above, for naming the individual interrupt lines

### Generic

Add an optional "parameter" to the data interpretation making the variable "push-type". The PCIE backend could then interpret that as "place to put my interrupt information into", but other back-ends could that just interpret it without the interrupt HW connotation. If demultiplexing of the variable is needed, the LNM plugin could be extended accordingly

killenb, 2019-08-22:
This is a very good proposal. I think we should do both, because they conceptually are a little bit different

### Push type

We have the field for the direction/access mode (ReadOnly, WriteOnly, ReadWrite). The read implicitly is a poll-type read. We could extend this by Push.
This information is not (or was not) used in the accessor creation, but just passed on to the application layer (for instance for ApplicationCore to determine the direction). That it is a push_type output fits well into this abstraction (to tell the application you can wait and don't need to trigger the read).

### Data interpretation

The data interpretation is information for the Backend how to convert the data. This is an implementation detail. One the application you only see what the resulting data type is, not where it is coming from (e.g. fixed point with fractional bits and IEEE754 both end up as `float` on the CPU (numeric, not integer in the catalogue)).
To describe that these data are interrupts. and how to interpret them, is on the same abstraction level. 

Note that the current proposal does not use the numerical addresses (address, size, bar), assuming that there is one interrupt mechanism, and the interrupts don't have data themselves (or one interrupt with a 32 bit field of data). The backend can use these to have different push-type variables at various addresses, which then contain data.

killenb, 2019-08-23:
Proposal/examples

1. One register with 32 bits signaling which interrupt is active. The application takes care to fiddle out which interrupt went on.
One has to select an address that does overlap with the other address spaces. I picked bar 0x9 (i for interrupt is the 9th letter in the alphabet).

```
# name           nElements  address  size    bar    width bit_interpretation signed mode
INTERRUPTS             0x1      0x0   0x4    0x9       32          INTERRUPT      0    P
```

2. Some push type registers in the "normal" address space. How to implement it is up to the backend. The current backends don't support this, but there might be future use cases.

```
# name         nElements  address  size    bar    width bit_interpretation signed mode
SOME_INTEGER         0x1       0x0   0x4    0x0      32                  0      1    P
SOME_FIXEDPOINT      0x1       0x0   0x4    0x0      16                  5      1    P
SOME_FLOAT           0x1       0x0   0x4    0x0      32            IEEE754      1    P
```
3. Void-Data interrupts which are individual bits from the actual interrupt bit (as @phako already mentioned this could be done in the logical name mapping, but I think it is clearer if it comes from the firmware, plus the dispatcher might be easier to implement and/or more efficient). The map
file can already contain meaningful names for the interrupts, not just I0, I1, I2 etc.

The additional mask in the bit interpretation selects the bit. In principle it would be possible to select multiple interrupts and mix then into a void, but I think it does not make sense. The firmware  could then already merge them as the same interrupt.
The data width is 0 (void data, just a trigger).
```
# name         nElements  address  size    bar    width bit_interpretation signed mode
DATA_READY         0x0       0x0   0x0    0x9        0       INTERRUPT:0x1      0    P
ERROR              0x0       0x0   0x0    0x9        0       INTERRUPT:0x2      0    P
LIMIT_EXCEEDED     0x0       0x0   0x0    0x9        0       INTERRUPT:0x4      0    P
```

mhier, 2019-09-09:
>     1. One register with 32 bits signaling which interrupt is active. The application takes care to fiddle out which interrupt went on.

This violates abstraction. Maybe not conceptually, but e.g. for PCIe we also have the DMA-ready interrupt, which will be part of this register then. I think we should not support this. I see no benefit, only potential future issues.

>     2. Some push type registers in the "normal" address space. How to implement it is up to the backend. The current backends don't support this, but there might be future use cases.

I would leave this out, too. Experience shows that specifying a future use case almost always goes wrong. We can add it later, if needed.

> 3. Void-Data interrupts which are individual bits from the actual interrupt bit (as @phako already mentioned this could be done in the logical name mapping, but I think it is clearer if it comes [...]

Actually, the map file kind of must specify meaningful names. We also do not have register names like "register0", "register1", etc. - this would defeat the (original) purpose of the map file. For interrupts this is no different.

> The additional mask in the bit interpretation selects the bit. In principle it would be possible to select multiple interrupts and mix then into a void, but I think it does not make sense.

It doesn't make sense and probably leads to undefined behavior resp. race conditions. Therefore I would change this into a plain number. The backend knows that this is a bit number of the interrupt bit field, it can convert this easily into a bit mask.

What might be missing in my eyes is something like a mixture of 2 and 3. Interrupts usually tell us when some data is ready to be read out. Ideally the map file would already mention *which* data this is. Hence we would need to attach an interrupt to a normal register. In the current scheme this would not be possible, but what if we would move the interrupt specification into the mode field (which currently is anyway redundant)? I mean like this:

```
# name         nElements  address  size    bar    width bit_interpretation signed mode
SOME_INTEGER         0x1       0x0   0x4    0x0      32                  0      1    INTERRUPT:1
SOME_FIXEDPOINT      0x1       0x0   0x4    0x0      16                  5      1    INTERRUPT:2
SOME_FLOAT           0x1       0x0   0x4    0x0      32            IEEE754      1    INTERRUPT:3
SOME_VOID            0x0       0x0   0x0    0x0       0                  0      0    INTERRUPT:4
```

killenb, 2019-09-09:
I agree on all three points. For 2 (regular push type registers) I was thinking about using it to simulate a pushing backend like the DOOCS backend with ZMQ subscription. But when implemented properly, the dummy has to be able to handle map files with interrupts and provide support for simulating interrupts.

About the bit masks I also agree. I wanted to make it similar to the bit interpretation in #74, but this is a different feature which can/should have different syntax.

I also like the idea of providing information about which data to read. But I would use it as an additional feature. I think there are use cases when there are interrupts which are not read triggers:
* Lukasz mentioned SPI_DONE, which would be the handshake after data has been written
* An ERROR interrupt which requires some other user action
In general, it is more flexible to have named registers defined which provide the interrupt.
In addition, I think it would be easier to implement this (not only in the dummy) if there are named accessors foreseen for each interrupt (but this is a side effect. Implementation should not affect the design decision unless absolutely necessary).

I could imagine something like this

```
# name         nElements  address  size    bar    width bit_interpretation signed mode

#define interrupts
DATA_READY           0x0       0x0   0x0    0x9       0   INTERRUPT:0x1      0    P
FLOAT_READY          0x0       0x0   0x0    0x9       0   INTERRUPT:0x2      0    P

#multiple registers read by DATA_READY interrupt (implicit definition of read mode)
SOME_INTEGER         0x1       0x0   0x4    0x0      32               0      1    INTERRUPT:DATA_READY
SOME_FIXEDPOINT      0x1       0x0   0x4    0x0      16               5      1    INTERRUPT:DATA_READY
SOME_FLOAT           0x1       0x0   0x4    0x0      32         IEEE754      1    INTERRUPT:FLOAT_READY
```

mhier, 2019-09-09:
I agree, mentioning the name of the interrupt instead of the number in the associated data register is more readable. Still I would make the same syntax for the non-data interrupts, as I had mentioned in my previous post (last line). Otherwise the syntax is a bit inconsistent for my taste.

```
# name         nElements  address  size    bar    width bit_interpretation signed mode

#define interrupts
DATA_READY           0x0       0x0   0x0    0x9       0               0      0    INTERRUPT:0x1
FLOAT_READY          0x0       0x0   0x0    0x9       0               0      0    INTERRUPT:0x2

#multiple registers read by DATA_READY interrupt (implicit definition of read mode)
SOME_INTEGER         0x1       0x0   0x4    0x0      32               0      1    INTERRUPT:DATA_READY
SOME_FIXEDPOINT      0x1       0x0   0x4    0x0      16               5      1    INTERRUPT:DATA_READY
SOME_FLOAT           0x1       0x0   0x4    0x0      32         IEEE754      1    INTERRUPT:FLOAT_READY
```

mhier, 2019-09-09:
Besides: I would also ignore the bar number and address in case of non-data interrupts. There is no additional information in putting a bar number 9 there.

killenb, 2019-09-09:
What I don't like now is that definition of interrupt and usage of interrupt are too similar. SOME_INTEGER a priory has nothing to do with the interrupt and could in principle be pull-read any time.

How about the following:
```
# name         nElements  address  size    bar    width bit_interpretation signed mode

#define interrupts
DATA_READY           0x0       0x0   0x0    0x0       0               0      0    INTERRUPT:0
FLOAT_READY          0x0       0x0   0x0    0x0       0               0      0    INTERRUPT:1

#multiple registers read by DATA_READY interrupt (implicit definition of read mode)
SOME_INTEGER         0x1       0x0   0x4    0x0      32               0      1    RO:DATA_READY
SOME_FIXEDPOINT      0x1       0x0   0x4    0x0      16               5      1    RO:DATA_READY
SOME_FLOAT           0x1       0x0   0x4    0x0      32         IEEE754      1    RO:FLOAT_READY
```
With this you have a clear separation of the void-type interrupts and the data registers that the application is supposed to pull. Obviously the device does this under the hood.
I think this is it conceptually not different from
```
# name         nElements  address  size    bar    width bit_interpretation signed mode
SOME_FIXEDPOINT      0x1      0x0   0x4    0x0      16               5      1    INTERRUPT:2
```
This would be more like a push-type register (my original example 2, which we said we don't have a use case for at the moment). Or maybe it's not that different? 

In any case, separating the two features allows a step-wise implementation.

(Notice that I changed from 0x1,0x2, 0x4 to the interrupt numbers 0,1,2 etc.)

mhier, 2019-09-09:
> What I don't like now is that definition of interrupt and usage of interrupt are too similar. SOME_INTEGER a priory has nothing to do with the interrupt and could in principle be pull-read any time.

This is one way to view it. Another is that both things are the very same, just one time you are defining a void register with interrupt, another time you define a non-void register with interrupt. Why shall this be a fundamentally different thing? Both registers will have their catalogue entries, simply with different data types. I think these two cases should really be as similar as possible.

Also I would not require defining interrupts separately from attaching them to data. This would be a restriction which might create problems for some protocol. The DOOCS backend is not a numeric addressed backend, but the push-type registers there would not fit into that scheme. I can imagine some numeric addressed backends might face the same problem.