ChimeraTK/DeviceAccess #86: Latest changes to ChimeraTK config file breaks DeviceAccess-Python

phako, 2019-09-28:
For some reason, the last commits I made declaring the boost dependencies publically causes the DeviceAccess Python bindings to segfault and miss symbols

phako, 2019-09-28:
https://github.com/ChimeraTK/DeviceAccess/commit/3925b5695b0f994fceaf84a143714b51c19ca033

killenb, 2020-01-14:
The problem seems that 
```list(APPEND ${PROJECT_NAME}_PUBLIC_DEPENDENCIES "Boost COMPONENTS system thread chrono")```
replaces the list of used boost libraries, not append to it. It seems to call the same mechanism as findPackage(boost). If you call 
```findPackage(boost COMPONENTS filesystem)```,
 and afterwards
 ```findPackage(ChimeraTK-DeviceAccess)```,
 the information for filesystem is no longer in the boost variables.

mhier, 2020-09-11:
@phako Is this issue still valid?

phako, 2020-09-11:
This should be fixed with https://github.com/ChimeraTK/DeviceAccess-PythonBindings/commit/9769b376507451cc0341cc4a45e8dbe1d7aa46fb

closed 2020-09-11 by phako