ChimeraTK/DeviceAccess #90: multiply plugin of the logical name mapping needs type conversion

killenb, 2019-11-25:
The multiply plugin needs an option to change the target type. (For instance multiplying an integer input with a fractional number needs to put out a floatingpoint number).

mhier, 2019-11-25:
With #89 this would not be required any more, one can use the conversion plugin first and then do the multiplication.

killenb, 2019-11-26:
Yes, this is already mentioned in #89. However, you have to create an additional variable (the double version of the integer that's coming from the device), which might be an unwanted copy. But I agree, if #89 is implemented, the priority for this ticket is low.

mhier, 2019-11-26:
No additional variable needed. We can have multiple plugins per variable.

killenb, 2019-11-26:
Ah, now I see. The converter plugin goes to the variable before the multiplier. This makes this ticket obsolete because we need the converter plugin anyway.

closed 2019-11-26 by killenb