ChimeraTK/DeviceAccess #97: Remove default implementation of isFunctional() in backends

killenb, 2020-01-14:
The default implementation of isFunctional, which always returns true one the device has been opened, is basically always wrong, since we have the requirement that the open is the logical state of "should be opened" and the backend tries to connect (in contrast to the device is closed and the backend does not try talk to the hardware).

Remove the default implementation from DeviceBackendImpl and see which Backends fails. First use a branch and make all backends work again by adding an implementation in each of them.

killenb, 2020-01-15:
For the PCIe and the Rebot backend dedicated implementations are needed for proper results. There are dedicated tickets for it. Merge the branch to the master only after #98 and #70 have been merged. Don't place implementations for PCIe and the Rebot backends, they will only create merge conflict. Just let the branch tests fail. (there whould be failing tests...)

nshehz, 2020-01-21:
Created follow up tickets #102 , #103 , #104 , #105

closed 2020-01-21 by nshehz