ChimeraTK/DeviceAccess #109: Rebot tests cannot be run in parallel

killenb, 2020-01-24:
The Rebot tests instantiate a dummy server, and all tests use the same port. Hence the tests cannot be run in parallel.

Use different ports. Best use random ports (check how it is done for doocs in the DOOC CS adapter and/or the DOOCS backend)

mhier, 2020-06-02:
Even worse, the tests will fail if run in parallel. Somehow, on Jenkins tests were not run in parallel recently due to a bug in our scripts there. Now this bug has been fixed and the Rebot tests fail consistently. We now have to force all tests of DeviceAccess to run sequentially because of this bug.

mhier, 2020-06-02:
Minimum solution would be to use a locking mechanism to prevent the tests from failing when running in parallel.

mhier, 2020-09-16:
See #176

commit e12b649b (2020-09-22): rebot tests: Use random server ports

Make it possible to run the tests in parallel

Fixes #109
closed 2020-09-22 by phako