ChimeraTK/DeviceAccess #116: readNonBlocking() and readLatest() don't always call postRead()

killenb, 2020-03-12:
readNonBlocking() and readLatest() only call postRead if there was new data. This is wrong. preRead() and postRead() must always be called in pairs, for instance if the user buffer needs to be restored.
In addidion, this will prevent proper exception propagation with the new exception handling scheme.

Solution in order not to perform unnecessary calculations:
change postRead() and doPostRead() to postRead(bool hasNewData) and doPostRead(bool hasNewData), and let the implementations decide what to do in case of no data.

killenb, 2020-04-03:
Fixed ApplicationCore ExceptionHandling Spec 11.23

killenb, 2020-04-06:
The "hasNewData" flag is the return value of "readTransferNonBlocking" or "readTransferLatest". For readTransfer use true.

For symmetry reasons the return value of write shall be added as postWrite(bool dataLost).

killenb, 2020-04-06:
Do before of after #119. It introduces an interface change at the same place.

ckampm, 2020-04-14:
> The "hasNewData" flag is the return value of "readTransferNonBlocking" or "readTransferLatest". For readTransfer use true.

This statement is contradicting with the second task in the opening comment. After `readTransfer`, the `hasSeenException` flag has to be tested in order to check if the transfer has been successful and provided new data.

ckampm, 2020-04-14:
,

closed 2020-04-14 by ckampm