ChimeraTK/DeviceAccess #117: Add readTransfer() etc.

killenb, 2020-04-03:
For all flavours of read and write transfers there currently are only the virtual doXxxTransferYyy functions, but not, like preXxx and postXxx non-virtual functions which do common functionality.

The doXxxTransfer functions might throw, because that's where the actual transfer can go wrong. 
... describe not decoratable due to transfer group... -> spec

Implement
* [ ] readTransfer()
* [ ] readTransferNonBlocking()
* [ ] readTransferLatest()
* [ ] readTransferAsync()
* [ ] writeTransfer()
* [ ] writeTransferDestructively()

such that they catch the exception thrown in the corresponding doXxxTransferYyy() function.

* [ ] They new functions are not virtual.

Introduce two member variables to TransferElement
* bool hasSeenException
* ChimeraTK::runtime_error lastException

* [ ] hasSeenException is set to false in preXXX
* if an exception is caught if xxxTransferYYY
   * [ ] hasSeenException is set to true
   * [ ] theException is stored in lastException
* [ ] if (hasSeenException) throw lastExcetion at then end of postXxx

mhier, 2020-04-08:
Question:

> if an exception is caught if xxxTransferYYY or doPostXxx is called

Where is doPostXxx called in these new functions?

mhier, 2020-04-08:
Comment about an implementation detail:

> hasSeenException is set to false in preXXX

I instead will set hasSeenException to false by construction and whenever an exception is thrown in postXXX(). preXXX() will get an assert to check that hasSeenException==false. This ensures that there are no exceptions lost due to bugs in the logic.

mhier, 2020-04-08:
> bool readTransfer()

This should be void, because doReadTransfer() also is void.

mhier, 2020-04-08:
> ChimeraTK::runtime_error lastException

I will rename this into "activeException", as it describes better the functionality.

killenb, 2020-04-08:
> > ChimeraTK::runtime_error lastException

> I will rename this into "activeException", as it describes better the functionality.
Does it? The exception stays there, even if it was thrown hours ago. It is not active any more after it has been throwm

mhier, 2020-04-08:
> > > ChimeraTK::runtime_error lastException
> 
> > I will rename this into "activeException", as it describes better the functionality.
> Does it? The exception stays there, even if it was thrown hours ago. It is not active any more after it has been throwm

The documenting comment will say it's only valid when hasSeenException == true. There is no purpose of accessing the exception later. The other name suggests basically it will *only* (or at least mainly) be accessed later (i.e. *after* postXXX). This is clearly not the case.

killenb, 2020-04-08:
> > bool readTransfer()
> 
> This should be void, because doReadTransfer() also is void.

But doReadTransfer has the exception channel to pass on information while readTransfer never throws. I have to pass the flag hasNewData to  postRead. I thought why should readTransfer not give me this information. Always putting it to true would be wrong.

killenb, 2020-04-08:
> Question:
> 
> > if an exception is caught if xxxTransferYYY or doPostXxx is called
> 
> Where is doPostXxx called in these new functions?

doPostRead() is inside of postRead(), where the exception must be caught, the book-keeping must be completed, and the exception re-thrown. portWrite I would have to check if there is bookkeeping remaining.

mhier, 2020-04-08:
Ok now I get what you mean... Sorry.

mhier, 2020-04-08:
Actually not really:
There can be only one runtime_error exception at a time, not one per layer. Hence if a call to doPostXxx() throws an exception, I cannot have an exception on my layer. What would I anyway do with the second exception? I can only throw one exception at a time. Hence there should be no try-catch-block necessary in the postXxx() functions.

killenb, 2020-04-08:
> Actually not really:
> There can be only one runtime_error exception at a time, not one per layer. Hence if a call to doPostXxx() throws an exception, I cannot have an exception on my layer. What [...]

If you don't have an exception in your layer you have to try-catch.
When you enter postRead there are two possible ways to see an exception:
1. There is one on my layer, then I don't have to try-catch (but I would not make that distinction)
2. I don't see the exception, but the lower level throws (it might be that the doPostRead I am calling is delegating to another postRead). This I have to catch.

I only will get one, that's correct. That's why it is sufficient to just store it. If I caught an exception I cannot have had one myself, so I will not overwrite an active exception.

mhier, 2020-04-08:
Why do I have to catch that? I don't have to do anything if I don't have an exception but there is one thrown a layer down.

mhier, 2020-04-08:
We are not talking about the implementations here, are we? This is only the TransferElement base class here. Or am I supposed to fix all implementations as well right away?

killenb, 2020-04-08:
> We are not talking about the implementations here, are we? This is only the TransferElement base class here. Or am I supposed to fix all implementations as well right away?

Yes, we are only talking about postRead() and postWrite() in TransferElement. There are no changes to the doPostRead() and doPostWrite() implementations at the moment.

killenb, 2020-04-08:
> Why do I have to catch that? I don't have to do anything if I don't have an exception but there is one thrown a layer down.

You have to catch because postRead does something after delegating to doPostRead, and this has to be executed even if there is an exception.

Current code
```
    void postRead() {
      if(!readTransactionInProgress) return;
      readTransactionInProgress = false;
      doPostRead();
      hasActiveFuture = false;
    }
```

What we need:
```
    void postRead() {
      if(!readTransactionInProgress) return;
      readTransactionInProgress = false;

      try{
         doPostRead();
      }catch(ChimeraTK::runtime_error & e){
        // when I am here there was not exception on this layer, so I don't accidentally overwrite it
        hasSeenException = true;
        activeException = e;
      } 

      hasActiveFuture = false;

      if (hasSeenException){
        // can be true because I just caught it, or the readTransfer on this layer did before
        hasSeenException=false;
        throw activeException;
      }
    }

```

mhier, 2020-04-09:
I think `hasActiveFuture = false;` can move before doPostRead(). Then nothing is left.

killenb, 2020-04-09:
> I think `hasActiveFuture = false;` can move before doPostRead(). Then nothing is left.

Was not sure this would be ok. But we are already in postRead, which is called by the future which is closing the transfer. You are right, this is the best solution.

killenb, 2020-04-09:
Please leave a comment in the code that a try-catch block will be needed in case we have to put something after the doPostRead later.

smarsching, 2020-04-09:
I think the handling of exceptions can be designed in an even nicer way by using an `exception_ptr`.

```
std::exception_ptr caughtException;
try {
  // Code that might throw.
} catch(...) {
  caughtException = std::current_exception();
}

// Code that should always run.

if (caughtException) {
  std::rethrow_exception(caughtException);
}
```

You can learn more about this [here](https://en.cppreference.com/w/cpp/error/exception_ptr).

This solution has the advantage that it always works, regardless of the type of the exception.

killenb, 2020-04-09:
The last tick mark with dataLost and recovery accessor actually is not part of DeviceAccess. It is only ApplicationCore. I will update the description. It should be true (data was lost) if the write transfer fails.

mhier, 2020-04-09:
@smarsching We are aware of std::exception_ptr. Since we have only one single type of exception we want to catch here there is not much of a difference.

smarsching, 2020-04-09:
@mhier Okay, I just wanted to make sure. I prefer to use `std::exception_ptr` because it will also work when the called code accidentally throws a different exception than the expected one (which can easily happen, e.g. when using the STL or just the `new` operator), but that’s a design decision to be made by you.

killenb, 2020-04-09:
@smarsching That's exactly the point here. We intentionally only want to handle ChimeraTK::runtime_error. All other exceptions should go through unhandled

smarsching, 2020-04-09:
@killenb Ah okay, thanks for explaining. 😄

mhier, 2020-04-09:
Actually, backends may not throw any other exception than ChimeraTK::runtime_error at this point. ChimeraTK::logic_error is the only other exception which may be thrown by a backend, but it must be thrown in the doPreXxx() function (only reason to throw is writing read-only or reading write-only, this should be checked already there). Throwing anything else is simply a bug (violation of the API). Debugging will be simpler, if exceptions are not unnecessarily caught and rethrown, hence better let the original exception through. We will need a follow-up ticket for this, because I could imagine not all backends do it like this.

mhier, 2020-04-09:
> The return value of readTransfer operations (which is 'hasNewData') is always false in case of an exception

I cannot fully comply to that rule. readTransfer() has no return value (void), readTransferAsync() has a TransferFuture as a return value. In both cases, false cannot be returned.

closed 2020-04-09 by mhier