ChimeraTK/DeviceAccess #121: doReadTransfer etc. must not be called any more

killenb, 2020-04-07:
Child of ChimeraTK/ApplicationCore#123
Depends on #117 

All code which is executing the read and write sub-steps separately must not call doXxxTransferYyy directly any more, but call xxxTransferYyy (e.g. readTransferLatest() instead of doReadTransferLatest()).

This concerns all decorator-like Accessor-Implementations, for instance (list not complete)
* LNMBackendBitAccessor
* LNMBackendChannelAccessor
* NDRegisterAccessorDecorator

The TransferGroup is done in a separate ticket (#120)

mhier, 2020-04-09:
Does it hurt if I fix this aspect in the TransferGroup already?

mhier, 2020-04-09:
Only found the problem in TransferGroup, all the others were already fixed in #117. Since the TransferGroup already has the follow-up ticket (#120) I think there is no follow up ticket to write.

closed 2020-04-09 by mhier