ChimeraTK/DeviceAccess #122: backends must not assign a fresh version number if the read transfer failed

killenb, 2020-04-07:
When a write transfer fails due to an exception, backends must not generate a new version number. They especially must not replace the nullptr of the "value after construction" with a valid version number.

Scan all known backends for places were version numbers are generated.
Known places:
* NumericAddressedLowLevelTransferElement::doPostRead()

mhier, 2020-04-15:
I think this just has been solved by #127

mhier, 2020-04-15:
I will check it.

mhier, 2020-04-15:
Which backends are the "known backends"? Apart from the DeviceAccess-internal backends I know only the DoocsBackend... Am I missing any other?

zenker, 2020-04-15:
E.g. OPC-UA backend and Modbus backend. At least in case of the modbus backend the version number is created by the NumericAdressed backend...

mhier, 2020-04-15:
> E.g. OPC-UA backend and Modbus backend. At least in case of the modbus backend the version number is created by the NumericAdressed backend...

I fear I will have to leave those to you... At least within the focus I cannot quickly deal with them as I have never compiled them before (and they presumably will have dependencies...).

mhier, 2020-04-15:
> Write a test that checks that the "value after construction" is not overwritten if the device is not available yet.

Not sure if the test will test what you expect. "Device not available" in DeviceAccess can only mean it is not opened (doesn't even matter why). In that case, any attempt to do any read or write on an accessor will result in a logic_error. This code path is not used in ApplicationCore, the ExceptionHandlingDecorator specifically has to prevent this.

mhier, 2020-04-15:
The "internal" part of this ticket is done, i.e. all backends within DeviceAccess are checked and tests are written. External backends (DoocsBackend) still have to be done.

killenb, 2020-04-16:
> The "internal" part of this ticket is done, i.e. all backends within DeviceAccess are checked and tests are written. External backends (DoocsBackend) still have to be done.

This is only about DeviceAccess. ControlSystemAdapter, ApplicationCore and DoocsBackend have their own tickets.
ChimeraTK/ApplicationCore#153
ChimeraTK/ControlSystemAdapter#26
ChimeraTK/DeviceAccess-DoocsBackend#27

The  DoocsBackend ticket might have to be refined.

mhier, 2020-04-16:
In that case this ticket is done.

closed 2020-04-16 by mhier

killenb, 2020-04-16:
> Not sure if the test will test what you expect. "Device not available" in DeviceAccess can only mean it is not opened (doesn't even matter why). In that case, any attempt to do any read or write on [...]

Well, "not available" can also mean that the device is opened, but isFunctional() reports false. You can reach this state with the exception dummy by letting it throw only on read. You can open it, and when you try to read the transfer will throw, which is delegated to the postRead. After this the version number must still be {nullptr}

There will be such a test in ApplicationCore, but only for the dummy. Don't know how to test this in general with any backend.

zenker, 2020-04-16:
So do you also want to create tickets for my backends?
Sorry I have not yet checked if anything needs to be done in those backends at all. 
I will only start working on them once your done with all other backends, i,e. the DOOCS backend, to see a working example...

mhier, 2020-04-16:
> Well, "not available" can also mean that the device is opened, but isFunctional() reports false. You can reach this state with the exception dummy by letting it throw only on read. You can open it, [...]

In which way is this different from a normal fail during read? isFunctional() is not used in this context, it plays no role.

There are only two possible failure modes to distinguish: a logic_error during and a runtime_error during read(). We don't care why the respective exception is thrown. It simply makes no difference in this context.

Btw: I am only testing that the value after construction stays unchanged under those exceptions. This is sufficient, because the value is no different from any other value that had been received in a previous read operation for this purpose.

> There will be such a test in ApplicationCore, but only for the dummy. Don't know how to test this in general with any backend.

ApplicationCore doesn't care about the backend, i.e. there you can sufficiently test with any single backend.

mhier, 2020-04-16:
> So do you also want to create tickets for my backends?
> Sorry I have not yet checked if anything needs to be done in those backends at all.

This is in principle up to you, but maybe it would be a good idea to create tickets so this doesn't get forgotten. I am pretty sure something will have to be done... (probably they won't even compile) :-)

> I will only start working on them once your done with all other backends, i,e. the DOOCS backend, to see a working example...

This is a good idea. It will help to identify what is to be done. Also we might introduce more generic tests in the UnifiedBackendTest, which you could apply to your backends to check if they are correctly implemented.

killenb, 2020-04-16:
> So do you also want to create tickets for my backends?

Maybe we should have one "umbrella" ticket which has the descripion, and then tickets in the individual backend projects to implement it. I think I created a 'stub' ticket for Modbus, but not for OPC UA. I will check.

@mhier is right, the backends currently don't even compile.

zenker, 2020-04-16:
My continuous integration did not even reach the point of testing the backends, since the stuff before does not compile any more (for me everything stuck with ContralSystemAdapter, which is part of the docker container I use to test my CTK software).

killenb, 2020-04-16:
> My continuous integration did not even reach the point of testing the backends, since the stuff before does not compile any more (for me everything stuck with ContralSystemAdapter, which is part of the docker container I use to test my CTK software).

Yes, we had to extend the interface of DeviceAccess. Backends should only depend on that. But for ApplicationCore and ControlSystemAdapter you also need the latest trunk. Otherwise it does not complile any more (neither your server not the ControlSystemAdapter or the ApplicationCore)