ChimeraTK/DeviceAccess #127: Check side effects of #116

killenb, 2020-04-14:
Child of ChimeraTK/ApplicationCore#123
Depends on #116 

Until now readLatest() and readNonBlocking() the postRead() function has not been called if there was no new data. This is wrong, there might be stuff to do like exception handling.
In #116 postRead() got a new argument `hasNewData`. This has to be used in each implementation to device if action is required if there is new data.

* [ ] Check each postRead() implementation if the action should always take place, or only if there is new data.
* [ ] Note in this ticket which places have been changed.

#122 is already dealing with the generation of new version numbers in NumericAddressedLowLevelTransferElement and LNMBackendVariableAccessor.

Examples
------------
* If preRead() is empty, and post read swaps the new data into the user buffer, this must not happen because the old user buffer must stay intact => no action if no new data
* If preRead() swaps out the user buffer such that the transfer can modify it, which it does not do because there was no new data, it has to be swapped back to get the user buffer back into place => action always required, even if no new data

mhier, 2020-04-15:
List of all `doPostRead()` implementation (via `grep "void *doPostRead" -R .`):

- [ ] ./device_backends/LogicalNameMapping/src/LNMMathPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/src/LNMMultiplierPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/src/LNMMonostableTriggerPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/include/LNMBackendChannelAccessor.h
- [ ] ./device_backends/LogicalNameMapping/include/LNMBackendBitAccessor.h
- [ ] ./device_backends/LogicalNameMapping/include/LNMBackendVariableAccessor.h
- [ ] ./device_backends/include/NDRegisterAccessorDecorator.h
- [ ] ./device_backends/include/TransferElement.h
- [ ] ./device_backends/include/NumericAddressedLowLevelTransferElement.h
- [ ] ./device_backends/include/NumericAddressedBackendMuxedRegisterAccessor.h
- [ ] ./device_backends/include/CopyRegisterDecorator.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/Subdevice/src/SubdeviceBackend.cchasNewData)
- [ ] ./device_backends/Subdevice/include/SubdeviceRegisterAccessor.h
- [ ] ./tests/executables_src/testTransferGroup.cpp
- [ ] ./tests/executables_src/testAsyncRead.cpp
- [ ] ./tests/executables_src/testDataConsistencyGroup.cpp

mhier, 2020-04-15:
I now realise, there are two aspects:
- prevent post-read action in case of an exception
- prevent post-read action in case of successful read transfer but without new data (exists only with AccessMode::wait_for_new_data)

The first aspect might be intrinsically covered if e.g. a delegated postRead() (without do) is called before the action, because that one will throw the exception and hence prevent the action. Still the second aspect need to be covered, unless the implementation never can have AccessMode::wait_for_new_data - but this is usually not the case for decorator-like implementations.

There might be additional aspects, also for write operations: Some actions might be necessary *despite* an exception is thrown in a delegated call. This is not covered by the ticket description. Is there a separate ticket for this?

mhier, 2020-04-15:
need to do this also for write (see my last comment). will do this as part of this ticket.

mhier, 2020-04-15:
List of all `doPostWrite()` implementations:

- [ ] ./device_backends/LogicalNameMapping/src/LNMForceReadOnlyPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/src/LNMMathPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/src/LNMMultiplierPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/src/LNMMonostableTriggerPlugin.cc
- [ ] ./device_backends/LogicalNameMapping/include/LNMBackendBitAccessor.h
- [ ] ./device_backends/include/NDRegisterAccessorDecorator.h
- [ ] ./device_backends/include/NDRegisterAccessorDecorator.h
- [ ] ./device_backends/include/TransferElement.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/include/NumericAddressedBackendRegisterAccessor.h
- [ ] ./device_backends/Subdevice/src/SubdeviceBackend.cc
- [ ] ./device_backends/Subdevice/include/SubdeviceRegisterAccessor.h
- [ ] ./tests/executables_src/testTransferGroup.cpp
- [ ] ./tests/executables_src/testAsyncRead.cpp
- [ ] ./tests/executables_src/testDataConsistencyGroup.cpp

mhier, 2020-04-15:
Btw: we can use a "finally" construct to avoid try-catch-blocks and code duplication. I have copied it from GSL to cppext (to avoid another dependency). Hence we need a new version of cppext.

mhier, 2020-04-15:
write part also checked and fixed.

closed 2020-04-15 by mhier