ChimeraTK/DeviceAccess #129: LNMBackendBitAccessor lock logic flawed

mhier, 2020-04-15:
The LNMBackendBitAccessor needs a lock to prevent concurrent manipulation of different bits in the same word. The lock is currently implemented with lock_guard in each steps of the transfer. This is incorrect, since the same target accessor is used by multiple LNMBackendBitAccessors. If concurrent transfers of different LNMBackendBitAccessor of the same word take place, the individual calls to preXxx, doXxxTransferYyy and postXxx are protected, but in between the calls of other transfers might take place.

With the new pre/post action logic, which guarantee us that they are always called in matching pairs, the logic can now correctly be implemented: in preXxx, the lock needs to be acquired, only in postXxx in must be released. This ensures that the entire transfer is protected against concurrent transfers.

mhier, 2020-05-28:
This can cause actual trouble on server start, see (sorry, DESY-internal link):
https://ttfinfo.desy.de/LLRFelog/show.jsp?dir=/2020/22&pos=/2020/22/2020

killenb, 2020-08-25:
@mhier Is this issue still present or has it been fixed as a side effect of the framework actions?

mhier, 2020-08-25:
I think it's still present, it only came up during the framework actions...

mhier, 2020-09-01:
Seems to be more complicated than originally through. Problem: Inside a TransferGroup, multiple operations on the target accessor are done at the same time but within the same thread. This results in a deadlock when using the naive implementation as done in 34cb73b7a9a057ec3cd58c4193040b52096433f7.

mhier, 2020-09-01:
fb08d9feffd618c973fac4ae536cb18a05ebcb52 tries to implement a more appropriate scheme, but it is quite complicated. Thread-local flags are required to determine whether a lock needs to be obtained or not. Is it now done correctly? Can this be done easier? Maybe someone should review this.

mhier, 2020-09-01:
Now on a second (or third, or fourth...???) thought, maybe a std::recursive_mutex should be used, this could solve the issues more elegantly?

killenb, 2020-09-01:
How about a recursive lock?

killenb, 2020-09-01:
Seems we had the same idea at the same time

phako, 2020-09-02:
Did check in call. Last recursive lock approach looks good

commit 34cb73b7 (2020-09-01): fix #129: correct broken locking scheme in LNM backend bit accessor

Also fix a potential race condition when obtaining the target accessor, in cases where the target accessor weak_ptr is expiring concurrently with the bit accessor's constructor.
closed 2020-09-02 by phako