ChimeraTK/DeviceAccess #144: TransferElement spec: version number and data validity need to be part of the application buffer

mhier, 2020-05-26:
TransferElement::getVersionNumber() and TransferElement::dataValidity() currently are virtual functions. Since the TransferElement spec says that this information in part of the application buffer, it should be treated in the same way as the NDRegisterAccessor::buffer_2D value buffer. Hence:

- add protected data members TransferElement::_versionNumber and TransferElement::_dataValidity
- make the two accessor functions non-virtual, the implementation just returns the two data members

To be decided: TransferElement::setDataValidity() probably should set the TransferElement::_dataValidity data member and hence should not be virtual, too.

mhier, 2020-05-26:
This ticket needs to be discussed. It is not strictly necessary to do it like this, but would help a lot against wrong handling of these application buffer elements in some cases (-> logical name mapper bit accessor). On the other hand, it makes some other decorator-like implementations a bit harder (and maybe slightly less efficient, although it saves some virtual function calls on the other hand, so it might level out), since the information has to be copied explicitly along with the data part of the application buffer. This goes wrong whenever the target accessor can be used without the particular decorator-like instance (this is exactly the case in the bit accessor). Maybe in some cases, an application might even want to use the same accessor with and without a decorator? Is that a legal use case?

mhier, 2020-05-26:
If we implement this ticket (maybe even otherwise), we should change the doPostWrite() interface to pass on the VersionNumber. This would greatly help implementing the write case correctly, since the application buffer should not be touched during writeTransfer(), doPreWrite() on the other hand is too early (writeTransfer() might fail, in which case the version number must not change).

killenb, 2020-05-27:
This would mean every decorator level has to copy the meta data as well, like with the data buffer. Is this more efficient than a virtual function call?

killenb, 2020-05-27:
> we should change the doPostWrite() interface to pass on the VersionNumber.

Well, we don't do it with the data buffer itself. The transfer copies to somewhere implementation-specific, and then we copy into the application buffer. Why should we pass it on for the version? I would not add another function argument. Maybe we already have too many.

killenb, 2020-05-27:
In principle I agree that we should treat all components of the application buffer equal. But as we have to touch all decorators (especially MetaDataPropagatingDecorator in ApplicationCore) I would first get the other functionality of exception handling and initial value propagation running and have more tests, to be sure we don't break anything.
It does not bring a high level functionality change, so it's not strictly needed for the next release (although introducing another API break in the next to next release again is also not nice).

mhier, 2020-05-27:
> This would mean every decorator level has to copy the meta data as well, like with the data buffer. Is this more efficient than a virtual function call?

If the meta data is actually accessed, yes. Accessing requires currently one virtual function call per layer. Copying one word of data is usually less expensive than a virtual function call. At least in ApplicationCore we always access all meta data. In the other cases, the performance impact should be negligible.

mhier, 2020-05-27:
To summarise our decision (in our phone call): The information is basically always needed in doPostWrite(), and it is very annoying if we have to store it in all implementations temporarily in a member variable. Hence we will do this interface change already now.

mhier, 2020-05-27:
We can actually take the burden of updating the version number in postWrite() from the backend implementation and put this into the TransferElement base class.  It always needs to be implemented in the very same way: if doPostWrite() does not (re)throw an exception, postWrite() will set the version number to the version number provided to the write() call. Hence there is no need to change the interface of doPostWrite().

mhier, 2020-06-02:
> Hence there is no need to change the interface of doPostWrite().

The interface change is still required, since decorators need to delegate to postWrite() from doPostWrite().

killenb, 2020-06-05:
* Implements spec B.4.3
* Take care of B.4.2.4: None of the transfers must directly touch the data members of the application buffer, incl. version number and data validity

killenb, 2020-06-17:
Implementation done. Closing the ticket.

closed 2020-06-17 by killenb