ChimeraTK/DeviceAccess #159: Write tests for Device::setException

killenb, 2020-06-15:
Probably part of the generic backend test

mhier, 2020-06-16:
The current assumption should be that everything stated in the TE specification is either tested by the TransferElement base class test or by the UnifiedBackendTest.

killenb, 2020-06-16:
How does the unified backend test check that all accessor implementations in the LNM Backend do it correctly?

mhier, 2020-06-16:
Generally, the UnifiedBackendTest is provided with a list of accessors which all are tested (actually multiple lists, to separate registers by capabilities, like sync and async readable and writeable registers). We of course have to make sure that all implementations are covered by these lists.

killenb, 2020-06-16:
ah, ok. sounds good.

mhier, 2020-07-24:
done as part of the UnifiedBackendTest

closed 2020-07-24 by mhier