ChimeraTK/DeviceAccess #173: testSharedDummyBackend is failing

killenb, 2020-08-17:
It seems the test is provoking an error condition, but not always catching it at the right place. An exception get's through, and at a different place the test is complaining that nothing has been caught. Probably a race condition that the exception usually
is thrown in two places, and only the typical place is being monitored.
```
##### Test capturing of starting supernumerous processes #####
Running 1 test case...
unknown location(0): fatal error in "testReadWrite": std::exception: Maximum number of accessing members reached.
/scratch/source/tests/unitTestsNotUnderCtest/testSharedDummyBackendExt.cpp(150): last checkpoint

*** 1 failure detected in test suite "SharedDummyBackendTest"
Testing PID Management of SharedDummyBackend failed. Starting a supernumerous process has not been catched.
Running 1 test case...
```

The bug probably is on the test.

killenb, 2020-08-25:
Showed in build #771, which has been locked.
It has not shown since, so I postpone the ticket now. Needs further observation if it re-occurs.

mhier, 2020-09-16:
Task for this ticket: check thoroughly if this issue still appears by running the test under various load conditions in a loop. If it doesn't appear any more, close this ticket. Otherwise try to fix the issue.

phako, 2020-09-21:
Looks like I can reproduce it

phako, 2020-09-21:
The one time I could reproduce it the whole SHM mechanism went haywire and the processes got stuck. I had to reboot the PC. Not happening since then

mhier, 2020-09-22:
Wow, even deleting *ChimeraTK* in /dev/shm didn't help then?

phako, 2020-09-22:
No but the PC was running for quite some time I don't think I can blame the test for that, more of a coincidence

phako, 2020-09-22:
Summing up: There are three likely causes for this.

1. This is a side-effect of #96 and the 11th process was stuck on shutdown
1. The parent shell did not wait long enough
1. The SHM in the container was in a bad state

3. can be excluded from the test logs, the SHM seems to work fine.

I have doubled the timeout to 10s in the test now. #96 needs definitely fixing anyway.

closed 2020-09-22 by phako