ChimeraTK/DeviceAccess #191: bad_numeric_cast exception: review code changes and spec

mhier, 2020-11-03:
Review the following commits:
* d3972b6c38621bc8f990b503d15e86d7ed8890e3
* 5e82804c106ba1de14cbb4f0700169bee8ad79f3
* 949ff290055ef5a3d0bf421173d9a5f6af2547e1
to make sure they are not (significantly) violating the spec.

They are somewhat contradicting the TransferElement spec C.4 (in particular 4.3) which still mentions the bad_numeric_cast exceptions. This part of the spec (and all other references to this exception) shall be reviewed as well. It should be decided which of the following solutions shall be implemented:
* overflows always result in the closest possible value
* asymmetric behaviour: in writes, overflows always result in the closest value, while for reading overflows are prevented in the constructor phase (logic_error if type is too small).

The second solution raises the question how to handle (as an extreme example) bidirectional variables between ApplicationModules involving a conversion (which is currently not implemented, but maybe it should be in future?).

killenb, 2020-11-05:
A discussion with @mhier lead to the conclusion that symmetric behaviour is the better solution.

* Overflows result in the closest possible value AND the data validity flag is set to faulty.
* Overflows in a single element of an array result in the maximum/minimum representable value only in that element. The data validity flag, which is for the whole process variable, is set to  faulty.

killenb, 2020-11-05:
I reviewed the code changes. They are fine.