ChimeraTK/DeviceAccess #193: Move to libxml++3 api (or use another XML library)

phako, 2020-11-13:


mhier, 2020-11-13:
Is that available on Ubuntu 20.04? I see only a libxml++2.6-dev package...

phako, 2020-11-13:
Apparently not. Which is odd, I think it is older than the Ubuntu LTS freeze. but ok.

mhier, 2020-11-13:
I think it is way older... I cannot find the source code repo for libxml++, the link on the webpage is dead. If it stays like this, we seriously should look for alternatives... Aren't there anyway more C++ish parsers available nowadays?

phako, 2020-11-13:
Latest release is from January: https://download.gnome.org/sources/libxml++/4.9/ There was some discussion moving the non-GNOME related cpp parts like sigc++ and xml++ somewhere else but I don't know where they went to. there's also 4.0 and 5.0 API. 

https://github.com/libxmlplusplus/libxmlplusplus is the correct one, https://libxmlplusplus.github.io/libxmlplusplus/ the correct page. I suspect the sf page still "belongs" to Murray

phako, 2022-10-18:
Still trying to talk to people, having libxml++5 api would be nice, since it removes a bit of dependency