ChimeraTK/DeviceAccess #204: Add test, that writes partial arrays, to the UnifiedBackendTest

zenker, 2021-03-26:
Writing partial arrays, where the Accessor size is different from the target array length, is not tested by the UnifiedBackendTest.
If there is some offset set and the array length is different from the Accessor size, one has to read the whole array once and only modify the elements accessed by the Accessor. This is not the case if partial writing is possible. 
It is not possible e.g. for the DOOCS, EPICS, OPC UA backend. 
The correct implementation should be tested to make sure backends where partial writing is not possible do everything correct (read_modify_write).

mhier, 2021-03-29:
Test for partial reads is also missing. This is not as easy to go wrong but still should be tested. In principle, the UnfiedBackendTest should test both partial reads and writes for any array register.

phako, 2025-03-07:
Implemented with ae216d92774324cd10e8f672189e155129f18ad9

closed 2025-03-07 by phako