ChimeraTK/DeviceAccess #233: IEEE754 for 2D registers with multiplexed data is not working

zenker, 2022-03-07:
Let's consider a multiplexed 2D register in a map file. 
```
APP0.AREA_MULTIPLEXED_SEQUENCE_DAQ0_BUF0                         500000            0      2000000     13     32      0    1   RO
APP0.SEQUENCE_DAQ0_BUF0_0                                             1            0            4     13     32      0    1   RO
APP0.SEQUENCE_DAQ0_BUF0_1                                             1            4            4     13     32      0    1   RO
APP0.SEQUENCE_DAQ0_BUF0_2                                             1            8            4     13     32      0    1   RO
APP0.SEQUENCE_DAQ0_BUF0_3                                             1           12            4     13     32      1    1   RO
```
I know that the register values are of type IEEE754. So I tried to specify that in the map file:
```
APP0.AREA_MULTIPLEXED_SEQUENCE_DAQ0_BUF0                         500000            0      2000000     13     32      IEEE754    1   RO
APP0.SEQUENCE_DAQ0_BUF0_0                                             1            0            4     13     32      IEEE754    1   RO
APP0.SEQUENCE_DAQ0_BUF0_1                                             1            4            4     13     32      IEEE754    1   RO
APP0.SEQUENCE_DAQ0_BUF0_2                                             1            8            4     13     32      IEEE754    1   RO
APP0.SEQUENCE_DAQ0_BUF0_3                                             1           12            4     13     32      IEEE754    1   RO
```
When I try to access such a register I get the exception: 
_Info: An exception was thrown:
Access mode flag 'raw' is not known by this backend._
The corresponding register called `MULTIPLEXED_RAW` contains only zeros although I know that there are values other than 0 in.
In my ApplicationModule I also see/receive only zeros.

killenb, 2022-03-07:
What has changed: You cannot get raw accessors from 2D registers any more. What I don't understand: You are not trying to do that, if I understand correctly. I will try to reproduce this.

killenb, 2022-03-07:
Actually you never could get working raw accessors of 2D registers, independently of the IEEE754. Maybe this is triggering a completely wrong exception message.

zenker, 2022-03-07:
By the way the problem was also present before the latest changes in DeviceAccess. And yes I don't try to access raw accessors. I simply would like to read `DAQ_BUF0`.

killenb, 2022-03-07:
There are two problems
a) The ` Access mode flag 'raw' is not known by this backend.` is probably coming from a wrong catalogue or meta data field. I can get a TwoDRegisterAccessor<float> without triggering the exception
b) The accessor is not working. It interprets the data as int, not IEEE754

killenb, 2022-03-08:
Fixed in the current master

closed 2022-03-08 by killenb

zenker, 2022-03-10:
Seems to work. 
Thanks for solving the issue immoderately.