ChimeraTK/DeviceAccess #256: Compilation fails in Yocto

cddf, 2022-05-05:
I created Yocto recipes for `cppext`, `exprtk` and `DeviceAccess` to compile it for a Zynq7000.
The compilation for `DeviceAccess` fails after around 50 % of the process, the other recipes work.

The only relevant log information I found was in `log.do_compile`:
```bash
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
ERROR: Execution of '/code/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/deviceaccess/git-r0/temp/run.do_compile
```

Some further compilation tests:
- The error is the same on `master` and `uio` branch.
- On the local x86 machine I can compile the `master` branch (`uio` fails for a different reason), but `ninja` is not used anywhere.
- On the Zynq I gave up with the compilation after ~30h with 40% progress.

Is there a way to avoid ninja in the cross compilation, i.e. some configuration variable?
And is the `uio` ready to be used on the FPGA?

The corresponding Bitbake recipe is here:

<details><summary>Bitbake recipe</summary>

```bitbake
SUMMARY = "ChimeraTK core library: Provide (client) access to hardware devices and other control system applications. "
HOMEPAGE = "https://github.com/ChimeraTK/DeviceAccess"
LICENSE = "GPLv3"
LIC_FILES_CHKSUM = "file://COPYING;md5=1ebbd3e34237af26da5dc08a4e440464"
BRANCH = "uio"
SRCREV = "${BRANCH}"
PR = "r0"

S = "${WORKDIR}/git"

DEPENDS += "boost cppext exprtk-interface libxml++"
RDEPENDS_${PN} += "cppext exprtk-interface libxml++"

inherit pkgconfig cmake 

SRC_URI = "git://github.com/ChimeraTK/DeviceAccess.git;protocol=https;branch=${BRANCH}"

FILES_${PN} += "${libdir}/* ${datadir}/**"
FILES_${PN}-dev = "${libdir}/* ${includedir}"
```

</details>

mbuechl, 2022-05-06:
I tried to replicate this with a cross toolchain for `aarch64-xilinx-linux` (ARM Cortex-A53) that I had exported from Yocto.

With the `uio` branch at ef5e8bc45d6fb3b1b9f107defa738c4817c1b73a I get the same error as @cddf does with `OECMAKE_GENERATOR = "Unix Makefiles"`.
With the `master` branch at bc83e08760cac371964aa13d8e232a11b3423df8 the build passes - maybe because of the 64 bit `aarch64` ISA instead of the 32 bit Cortex-A9 armv7 that @cddf uses?

killenb, 2022-05-06:
The `uio` branch is old and broken at the moment. I will check what we will do about it later today. 
Most of the new functionality in that branch is already in the master since version 02.06 (AsyncRegisterAccessor)

The uio branch was created based on version 01.04. We probably will re-implement the existing uio functionality based on the current master (03.01), since the existing branch is two major revisions behind, and re-basing failed. That's why the branch is broken at the moment.

phako, 2022-11-25:
I can reproduce the arm32 issue with a plain yocto

phako, 2022-12-06:
@cddf please try our layer from https://github.com/ChimeraTK/meta-chimeratk - this should fix your issues.

phako, 2022-12-12:
And there should be unio available as of now as well.

closed 2022-12-12 by phako

cddf, 2022-12-15:
Nice, that compiles, too. I'll try to test uio, soon.