ChimeraTK/DeviceAccess #375: Interrupt specifier should be independent of access modifier

patrislav1, 2023-11-27:
The property of a register being associated with an interrupt is independent ("orthogonal") of it being a RW, RO, or WO register. We should be able to have read-only interrupt registers as well as writable ones.

**Proposal**

* Make [NumericAddressedRegisterCatalogue::Access](https://github.com/ChimeraTK/DeviceAccess/blob/1e1e28745703db938b06a752b4be41b74b52fcc6/device_backends/NumericAddressedBackend/include/NumericAddressedRegisterCatalogue.h#L24C26-L24C26) a bit field (flags could be like "readable, writable, interrupt")
* Adapt [NumericAddressedRegisterInfo::isReadable/isWritable()](https://github.com/ChimeraTK/DeviceAccess/blob/1e1e28745703db938b06a752b4be41b74b52fcc6/device_backends/NumericAddressedBackend/include/NumericAddressedRegisterCatalogue.h#L85) accordingly
* Adapt the map file parser so that statements like `RW,INTERRUPT0` or similar are supported

mhier, 2023-11-27:
Curious, what is the use case of a register that is writeable and has an interrupt?

Normally, RW is understood as a "special" variant of WO which allows to read back the latest written value, but the firmware will never change the value of a RW register. Interrupts imply that the firmware changes the value on its own action. It will be difficult to make truly bi-directional registers free of race conditions, and I see no benefit in doing so.

patrislav1, 2023-11-27:
Typical use case would be a interrupt status/acknowledge register. Read for interrupt status, write for acknowledge. A bunch of Xilinx cores use a write-to-clear logic, i.e. write a "1" to clear the bit.

killenb, 2023-11-28:
I discussed the same with Lukasz yesterday, and made a ticket in our redmine: https://redmine.msktools.desy.de/issues/12496

killenb, 2023-11-28:
The catalogue does not necessarily need modification. The isReadable/isWritable flags are independent, and the AccessMode wait_for_new_data is turned on for interrupts, which works in all combinations.

patrislav1, 2024-07-16:
Now that interrupt handling is implemented, this can be closed.

closed 2024-07-16 by patrislav1